TheAntTeam/TheAntFarm #1: No module named gerber

dreschpe, 2022-01-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows, Ubuntu]
 - Version [e.g. 10, 18.04 LTS]

**Additional context**  
Add application log files from the folder *app_logs*.   

**Additional context**
Add any other context about the problem here.

dreschpe, 2022-01-06:
After installing I had to add two missing modules: PySide2 and viSpy. 
If I start I get the error: No module named 'gerber'

TheAntTeam, 2022-01-06:
Hi @dreschpe, do you have installed all the requirements as described in the [installation](https://github.com/TheAntTeam/TheAntFarm#installation) procedure? Gerber is a module of pcb-tool, one of the python packages required by TheAntFarm and it is present in the requirements.txt file.

dreschpe, 2022-01-06:
After re - installing all modules from the requirement.txt manually the software starts :-)

closed 2022-01-06 by dreschpe