TheAntTeam/TheAntFarm #2: Gerber File not loaded

dreschpe, 2022-01-06:
After starting the software I have found a Gerber File that is not rendered.
There is an error in the status line 
AttributeError: 'Outline' object has no attribute 'sides'

The layers are loading in flatcam without problems.
I will attach the files.

TheAntTeam, 2022-01-07:
Hi Peter,
we have checked our code and it seems that we have solved the problem.
Can you try our last program version (67b22b345d2c10a2818517018940dc5de70bc1db)? (just make a pull from the git repository)

Gerber files provided by you will enter our regression test list.
Thank you for your feedback!!!

TheAntTeam, 2022-01-11:
Closing since we received feedback that the problem is solved.

closed 2022-01-11 by TheAntTeam