TheAntTeam/TheAntFarm #4: Enabling the Milling Tool on the Drill job makes things stop processing.

bvarner, 2022-01-26:
Not sure if this is supposed to work yet or not, but it's about the only thing keeping me from using this tool to generate the g-code for boards. I like this tool _much_ better than flatcam, so far. Great job.

**Describe the bug**
I've setup several drill-bits and sizes based on my design rules.
I use 0.7mm for vias, 1.0 for pin headers, and have some other larger holes all > 1.0mm. 
For those larger cutouts, I like to use a 0.8mm endmill.

If I don't check the 'Milling Tool' checkbox, I see paths that visit each hole definition, and appear to select the 'best match' tool size for the hole based on debug output. Here's the output from the command line, and the visualization on screen:

```

Collect Geom
--- 0.0001239776611328125 seconds ---
Geom Collected
--- 0.012621164321899414 seconds ---
Final Merging
--- 0.01583242416381836 seconds ---
Cfg
OrderedDict([('bits_names', ['Bit6', 'Bit7', 'Bit8']), ('bits_diameter', [0.8, 0.7, 1.0]), ('milling_tool', False), ('tool_diameter', 0.8), ('cut', -1.2), ('travel', 1.0), ('spindle', 1000.0), ('xy_feedrate', 250.0), ('z_feedrate', 40.0), ('optimize', True)])
Drilling
Initial distance: 1716.108591508307
10.0
20.0
30.0
40.0
50.0
60.0
70.0
80.0
90.0
Final distance: 367.89052373329906
Bit 1.0 [(19.203999999769888, -19.041999999718133), (18.06499999977708, -28.87999999976711), (22.064999999777076, -28.87999999976711), (29.153999999659042, -48.041999999763036), (31.693999999668463, -48.04199999976304), (34.233999999705716, -48.04199999976304), (36.77399999974296, -48.04199999976304), (41.853999999817475, -48.04199999976304), (44.393999999854714, -48.04199999976304), (46.933999999864135, -48.04199999976304), (49.473999999853454, -48.04199999976302), (52.013999999890714, -48.04199999976302), (67.50799999981645, -62.991999999821694), (77.20399999976992, -68.04199999971819), (67.50799999981645, -68.9919999998217), (54.4689999997114, -68.9919999998217), (54.4689999997114, -62.991999999821715), (41.43099999977961, -62.99199999982169), (41.43099999977961, -68.9919999998217), (19.203999999769888, -68.04199999971819), (28.391999999759715, -68.99199999982172), (28.391999999759715, -62.99199999982171), (39.31399999978022, -48.04199999976304), (51.7599999997765, -34.02799999979657), (51.7599999997765, -31.48799999975932), (51.7599999997765, -28.94799999972207), (51.7599999997765, -26.407999999731153), (52.013999999890714, -20.311999999781666), (62.17399999964984, -20.311999999781676), (67.25399999972433, -17.77199999974442), (69.79399999976161, -17.77199999974442), (72.33399999979883, -17.77199999974442), (77.20399999976992, -19.041999999718133), (72.33399999979883, -20.311999999781676), (69.79399999976161, -20.311999999781676), (67.25399999972433, -20.311999999781676), (64.71399999968709, -20.311999999781676), (64.71399999968709, -17.77199999974442), (62.17399999964984, -17.77199999974442), (59.633999999688356, -20.311999999781662), (57.09399999965108, -20.311999999781662), (54.553999999672804, -20.311999999781662), (59.633999999688356, -17.771999999744416), (57.09399999965108, -17.771999999744416), (54.553999999672804, -17.771999999744413), (52.013999999890714, -17.771999999744413), (49.473999999853454, -20.311999999781666), (49.473999999853454, -17.771999999744413), (41.853999999817475, -20.311999999781676), (36.77399999974296, -17.77199999974442), (39.31399999978022, -17.77199999974442), (41.853999999817475, -17.77199999974442), (44.393999999854714, -17.77199999974442), (46.933999999864135, -20.311999999781676), (46.933999999864135, -17.77199999974442), (44.393999999854714, -20.311999999781676), (39.31399999978022, -20.311999999781676), (36.77399999974296, -20.311999999781676), (34.233999999705716, -17.77199999974442), (34.233999999705716, -20.311999999781676), (31.693999999668463, -20.311999999781676), (31.693999999668463, -17.77199999974442), (29.153999999659042, -17.771999999744413), (29.153999999659042, -20.31199999978167), (26.613999999669705, -20.311999999781662), (26.613999999669705, -17.771999999744416), (24.073999999632456, -17.771999999744416), (24.073999999632456, -20.311999999781662)]
Initial distance: 229.54268456598408
10.0
20.0
30.0
40.0
50.0
60.0
70.0
80.0
90.0
Final distance: 127.42913882727217
Bit 0.7 [(35.63299999977538, -22.198999999705748), (38.04399999976158, -22.343999999724378), (44.781999999657266, -27.289999999720653), (43.63199999975042, -28.439999999795052), (63.69799999985843, -52.823999999705755), (63.69799999985843, -55.36399999974301), (49.21999999973923, -52.823999999705755), (49.21999999973923, -55.36399999974301), (36.51999999970203, -55.363999999743), (36.51999999970203, -52.82399999970574), (23.311999999779097, -52.82399999970575), (31.69399999966845, -33.01199999973178), (36.26599999982863, -28.947999999858443), (36.26599999982863, -27.677999999839813), (34.995999999810024, -25.64599999971687), (36.062999999755235, -24.57899999977165)]
Writing gcode file:
	 /home/bvarner/Documents/Personal/TheAntFarm/gcode_temp_dir/drill_drill.gcode
Done
```
[image]

If I check the box for the 0.8mm Milling Tool....
Only the toolpath for the 0.7mm bit is shown.
```
Collect Geom
--- 0.0001068115234375 seconds ---
Geom Collected
--- 0.059865474700927734 seconds ---
Final Merging
--- 0.015403270721435547 seconds ---
Cfg
OrderedDict([('bits_names', ['Bit6', 'Bit7', 'Bit8']), ('bits_diameter', [0.8, 0.7, 1.0]), ('milling_tool', True), ('tool_diameter', 0.8), ('cut', -1.2), ('travel', 1.0), ('spindle', 1000.0), ('xy_feedrate', 250.0), ('z_feedrate', 40.0), ('optimize', True)])
Pocketing
Path Generation Done in 0.00532984733581543 sec
Drilling
Initial distance: 246.11572564544008
10.0
20.0
30.0
40.0
50.0
60.0
70.0
80.0
90.0
Final distance: 127.42913882727217
Bit 0.7 [(36.062999999755235, -24.57899999977165), (34.995999999810024, -25.64599999971687), (36.26599999982863, -27.677999999839813), (36.26599999982863, -28.947999999858443), (31.69399999966845, -33.01199999973178), (23.311999999779097, -52.82399999970575), (36.51999999970203, -52.82399999970574), (36.51999999970203, -55.363999999743), (49.21999999973923, -55.36399999974301), (49.21999999973923, -52.823999999705755), (63.69799999985843, -55.36399999974301), (63.69799999985843, -52.823999999705755), (43.63199999975042, -28.439999999795052), (44.781999999657266, -27.289999999720653), (38.04399999976158, -22.343999999724378), (35.63299999977538, -22.198999999705748)]
Writing gcode file:
	 /home/bvarner/Documents/Personal/TheAntFarm/gcode_temp_dir/drill_drill.gcode
Done
```
[image]

Here is my drill file (zipped):
[powerpiHAT-drl.zip](https://github.com/TheAntTeam/TheAntFarm/files/7946135/powerpiHAT-drl.zip)

TheAntTeam, 2022-01-29:
Hi @bvarner, in commit 22350f38dd0bd021e73905ed49c162494a5c056b you can find a solution for this issue.
Now all the paths are calculated correctly, shown in the canvas and reported in the gcode file with the relative tool change commands.
The pocketing path, if present, is always the first that will be executed during the drilling phase.

bvarner, 2022-01-29:
This is certainly an improvement! Thanks.
Should I be concerned that I don't see travel paths plotted between the pocketing paths?
[image]

Here's my output: I'm only seeing traveling optimization once in the output with the milling tool selected.
Without milling, I see all three bits have paths planned.

I have not inspected the gcode yet.

```
Collect Geom
--- 9.465217590332031e-05 seconds ---
Geom Collected
--- 0.01344609260559082 seconds ---
Final Merging
--- 0.015114545822143555 seconds ---
Cfg
OrderedDict([('bits_names', ['bit0', 'Bit1', 'Bit2']), ('bits_diameter', [0.7, 0.8, 1.2]), ('milling_tool', True), ('tool_diameter', 0.8), ('cut', -1.2), ('travel', 1.0), ('spindle', 1000.0), ('xy_feedrate', 250.0), ('z_feedrate', 40.0), ('optimize', True)])
Pocketing
Path Generation Done in 0.005555629730224609 sec
Drilling
Initial distance: 221.29545852697314
10.0
20.0
30.0
40.0
50.0
60.0
70.0
80.0
90.0
Final distance: 127.42913882727217
Path Generation Done in 5.570298194885254 sec
Compute Drill:
 - Pocketing, bit diameter 0.8
 - Drilling, bit diameter 0.7
Writing gcode file:
	 TheAntFarm/gcode_temp_dir/drill_drill.gcode
Done
```

TheAntTeam, 2022-01-30:
What you see is correct. In the paths view, one tends to exclude the travel paths (except for the drilling phase, otherwise you would not see anything).
Therefore, in this view, for the milling phases (gerber, profile and drilling pocket) we have chosen not to display the movement trajectories.
In this way we think they are easier to analyze to understand if everything is correct. If you upload the gcode file in the gcode section of the second tab, you will see the path in its entirety.

closed 2022-01-30 by TheAntTeam