TheAntTeam/TheAntFarm #11: Timeouts when sending GCode.

bvarner, 2022-02-21:
**Describe the bug**
When sending gcode files

The machine starts to follow gcode instructions, but after a fairly short period of time (less than a single pad routed) I get a message that the machine is disconnected, and TheAntFarm looses connection.

I can reconnect from the GUI, but then the machine stops, and it's in an Alarm state.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Control, select a file.
2. Click Run.
3. Wait a few seconds...
4. See error

**Expected behavior**
That the machine would be able to execute all the instructions in the gcode file. :-)

**Screenshots**
I didn't take one. It takes me a bit of effort to reproduce (ant controller is air gaped)

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: 21.10
 
 **Additional Context**
 I am using a modified grbl1.1 on an Arduino Uno... https://github.com/bvarner/varnerized_ant_grbl/tree/feature/varnerized_ant

TheAntTeam, 2022-03-03:
Hi @bvarner,
I tried to investigate and it seems that effectively on Ubuntu the serial resource sometimes becomes unavailable for some moments, I still don't know the reason of this. 
Effectively I had added a control that when the serial resource was not available anymore (e.g. someone disconnected the cable) the serial would be closed. 

This management is too drastic and when there is a momentary unavailability it causes the total loss of connection. 
So I commented out the closing of the serial port, while leaving the error logging. 

And I will try to search the reason of the momentarily unavailability of the port.
If this solves it also for you I will close it. Thank you :)

TheAntTeam, 2022-03-05:
I think the reason for the serial momentary unavailability could be what is explained here:

[python-serial-oserror-errno-11-resource-temporarily-unavailable](https://stackoverflow.com/questions/36133620/python-serial-oserror-errno-11-resource-temporarily-unavailable)

and in other links related.

I tried to disable the serial-getty service on my serial port like the suggestions said, but it did not change much for my ubuntu system. But maybe I did something wrong or I did not take something in consideration, I don't know right now.

closed 2022-04-04 by TheAntTeam