TheAntTeam/TheAntFarm #12: GCode Generated for bottom layer assumes a top-down view.

bvarner, 2022-02-21:
**Describe the bug**
Generating GCode for a bottom copper layer isn't something I can send to the ANT without modifying it.

**To Reproduce**
Steps to reproduce the behavior:
1. Import a bottom layer gerber.
2. Generate GCode.

Origin for the GCode is still in the bottom left. Flipping to the 'bottom view' would be the appropriate paths to send to the printer, but that's not what is reflected in the gcode.

**Expected behavior**
I would expect the bottom layer gcode to be generated mirrored, so that when you send the gcode to the ANT, it can mill a board which has been flipped upside down. Or for something to 'mirror' the gcode or the geometry prior to generating the gcode in a manner that it can be used.

dreschpe, 2022-04-14:
run into the same problem.

TheAntTeam, 2022-04-14:
Hi, thanks for your notification.
We added the mirroring flags for profile and drilling while for bottom it is implicitly always mirrored.

We will add also the possibility to set the mirroring axis, X or Y, in the Settings/Preferences tab.

bvarner, 2022-06-13:
I'm going to close this issue, as I've tested this now. :-)

Loading the GCode to send to the printer works, despite what the preview is showing now. Thanks!

closed 2022-06-13 by bvarner