TheAntTeam/TheAntFarm #15: Icon Grafik is not visible

dreschpe, 2022-04-14:
**Describe the bug**
I run TheAntFarm with windows11, Python 3.9.12
After start, the program is working, but all button graphics are not shown.
Only the text.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

TheAntTeam, 2022-04-14:
Hi,
are you launching the python main script from the project folder?

dreschpe, 2022-04-15:
Yes, I was launching it from the project folder.
I have switched to the second option with visual code and there I can see the graphics :-)

TheAntTeam, 2022-05-12:
Hello @dreschpe,
is the issue solved with the commit here above? If so, I would close the issue, otherwise we investigate further :)

dreschpe, 2022-05-12:
Unfortunately, the bug is still there.
I pull the actual code and follow the steps.
If I start it inside visual code it is working.

TheAntTeam, 2022-05-14:
Hi, did you activate the virtual env before starting the program? We are trying to reproduce the issue, but we are not able up to now.

dreschpe, 2022-05-19:
Yes I'm following all steps ...

Python 3.9.13
Win 11 Pro

TheAntTeam, 2022-05-19:
Ok, thank you, I was guessing that there was the resource path not correctly set for some reason, that's why I asked about the venv.  
  
If you can do it, one more test would be to re-build the resources.   
This can be done, if you have the pyside2-tools installed. on the terminal with this command:  

> .\build.py ui  

or with the following command:

pyside2-rcc app_resources.qrc -o app_resources_rc.py  
  
In the meanwhile we try to set up a window 11 machine to reproduce all the conditions.

alexlinno, 2022-05-22:
Hi I have the same issue, even running it from VS code. I see that I'm in venv and tried to rebuild (both ways) but still have the problem
W10 and Python 3,10

alexlinno, 2022-05-22:
after deinstalling and installing it all again I got the icons:-)

TheAntTeam, 2022-05-23:
Hi @alexlinno,
nice to know👍 
When you say "deinstalling and installing it all again" you mean that you re-did the whole installation process?
Do you know what could be different respect from the first installation?

dreschpe, 2022-08-12:
The bug is back, also with VS Code.
I can re-build the resources. It works. I have renamed one of the graphics and it fails.
The new file is built. The only graphic used by the program is the logo. 
Also, the newly built file is used. If I kill it the program does not start.
No idea ...

TheAntTeam, 2022-08-13:
>  I have renamed one of the graphics and it fails. The new file is built. The only graphic used by the program is the logo. Also, the newly built file is used. If I kill it the program does not start. No idea ...

Hi @dreschpe.
The "app_resources.qrc" file contains the lists of relative paths to the files used, if you changed the name or the relative path of a file also this file should be updated and then re-built into the "app_resources_rc.py" file.
Let us know if this solves the issue.

dreschpe, 2022-08-16:
The logo is a png file and is displayed. I can edit it and it is loaded.
The other graphics are svg files and they are not displayed.
Is there a problem with the component which is displaying the svg's ?
I'm no QT programmer, but I think if the path to the logo is working, also the path to the icons will work.
.