TheAntTeam/TheAntFarm #17: Cant Load Drill Files on Mac

nebulorum, 2022-04-18:
**Describe the bug**

I just did a test (still no machine) on the software to compare to FlatCam. First off while installing on the Mac worked fine, I failed to run because of the following error:

```
Linux Env
WARNING: Could not find the Qt platform plugin "xcb"
```

However with there following change it starts correct and I was able to load front, back and profile.
```diff
--- a/the_ant_farm.py
+++ b/the_ant_farm.py
@@ -28,7 +28,7 @@ else:
     print("Linux Env")
     os.environ["QT_PLUGIN_PATH"] = os.path.join(pys2_path, "plugins")
     os.environ["QT_QPA_PLATFORM_PLUGIN_PATH"] = os.path.join(pys2_path, "plugins", "platforms")
-    os.environ["QT_QPA_PLATFORM"] = "xcb"
+    #os.environ["QT_QPA_PLATFORM"] = "xcb"
```

Drill files fail to load, although I managed to get them on FlatCAM. 

**To Reproduce**
Steps to reproduce the behavior:
1. Patch file to allow Mac to load Cocoa;
2. Load Gerbels;
3. Select drill file (both PTH, and NPTH failed);
4. No file shows up.

There was also a strange thing on the preview table where I would see path for both sides after clicking on the different radio buttons for the layer I had. If this is not expected I can try to debug this a bit more, just need some guidance.

**Expected behavior**

Drill file gets loaded.

**Desktop (please complete the following information):**
 - OS: Mac OS 
 - Version: 10.15.5

TheAntTeam, 2022-04-25:
Hi @nebulorum,
could you send us the drilling files that you are unable to load?
About the preview table problem and the path loading, could you be more specific? Maybe with a step-by-step example?
Unfortunately we don't have a Mac to test on, so it is possible that we will ask you for some debug tests to run on your Mac.

nebulorum, 2022-04-27:
Here is the files [drill_files.zip](https://github.com/TheAntTeam/TheAntFarm/files/8577365/drill_files.zip). This was generated with Kicad 4. I have updated KiCad so could try to generate again. All the GBRL files loaded fine. Concerning debugging I'm willing to help. I'm still learning how to use the machine (bought a used one, which still needs to be updated).

nebulorum, 2022-05-17:
I did a quick test with the X2 files. Unfortunately it seems KICAD can't generate unified files here. But the file did load correctly. However the file picker was not really me to see that the file was eligible (all files were grayed out).

I also notices that some times older paths show in the UI even if they are not milled.

nebulorum, 2022-06-06:
While unrelated to the original issue, I've create a PR to make Mac work. #21

TheAntTeam, 2022-09-03:
We have made a commit (https://github.com/TheAntTeam/TheAntFarm/commit/65c55430274a49422607ab0bc9743358a1168644) which contains a possible solution to this issue. Could you verify that it also works in your environment?

Moorviper, 2022-09-05:
Looks good for me.
[image]
[image]
[image]