TheAntTeam/TheAntFarm #19: Drill Killer

dreschpe, 2022-05-19:
The actual version has a problem with the drill file or the drill start. 
I test a single side PCB. top layer, drill and outline are generated without mirroring.
The top layer is milled without problems.
After a tool change, the drill bit is leveled.
The drill is pushed at 0,0 without stopping into the board if I start the drilling.
The drill g-code looks ok.

Other problem :
If I switch between the different g-codes, the screen is not cleared.  Both working paths are shown. 

Peter

TheAntTeam, 2022-05-19:
Hi Peter,
could you give us the gcode files? Also by mail, if you want.

Are you executing the semi automatic tool change procedure like in [here](https://youtu.be/VqssE8qKiTY) ?  

If you are doing the tool change manually, are you zeroing the new tool Z?

bvarner, 2022-06-15:
I'm having issues with the drill files and tool change procedure process as well. 
I think it might be part of my GRBL settings.

Once I do a soft reset, my machine state is in 'alarm', so I have to `$X` to clear it.

I'm also having issues with the spindle turning on at the right time, but it may be my ESC.
I've normally had to do:

```
M3S10
M3S1000
M3S10
```

And then M3S<actual target> to get my ESC working.