TheAntTeam/TheAntFarm #20: Plot of deselect GCode not cleared

alexlinno, 2022-05-22:
When having to GCode files loaded selecting one and then the other shows both in the viewport.

alexlinno, 2022-05-22:
Also I cannot remove GCode fiels from the list

TheAntTeam, 2022-05-23:
Hi Alex,
thanks for opening this issue.
Could you send us the files you are using? 

If you can also list the packages you have installed in you virtual environment are report them here. The following command should be enough:  
  
> pip list  

Is there a specific sequence of actions that reproduces the issue always?

About removing the gcodes, the rows of the table are selected pressing the numbers, you may select multiple rows and then remove them.

TheAntTeam, 2022-08-20:
I think the commit 1d6133ad463514815fe7aae19ba247b4e967993b solved the issue.
If it is not so, we will re-open it.

closed 2022-08-20 by TheAntTeam