TheAntTeam/TheAntFarm #23: Bottom layer ABL Errors

bvarner, 2022-06-14:
**Describe the bug**
Any time I'm milling a bottom layer (have not tried on top-layer -- will do so later this evening) ABL applies z-coordinates that end up in the positive z-axis range in the top right corner of the board. (See attached photo)

You can see the way the tool is lifting in this photo.
[image]

**To Reproduce**
Steps to reproduce the behavior:
Load a bottom gerber, generate a job.
Load the gcode into the control tab. Bottom layer gcode is properly displayed, and the job looks good.
Switch to the sender, connect to machine, set work pos offset on X-Y = 0.
Click "Get BBox", make sure ABL checkbox is selected.
Set ABL step sizes.
Click "ABL"... (I then wait until the finished message shows up in the console.)

Send the GCODE to the ANT.

As the cutter approaches the top-right corner of the board, I consistently get Z WPos in the positive range.
I'm not certain yet, but I think it may be cutting more deeply than I've had the job setup specified to cut.

**Expected behavior**
I expect it not to lift into positive Z axis working space after ABL, when the board clearly warps into the negative space.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version Several releases. Running on python3, in venv.

**Additional context**  
Add application log files from the folder *app_logs*.   

**Additional context**
I plan to re-test this by mirroring the plots from kicad, and generating a set of files I can mill as a 'top' layer instead. I have a sneaking suspicion this has something to do with the 'flip' of the coordinates, but that's just an unproven hun

bvarner, 2022-06-15:
Can confirm: Top Layers mill isolation routes perfectly.

TheAntTeam, 2022-08-22:
Ok, the behavior is replicable. Debug in progress

TheAntTeam, 2022-08-23:
Hi, can you try the fix? I have done some tests and It seems working

closed 2022-09-03 by TheAntTeam