TheAntTeam/TheAntFarm #27: ABL not working

dreschpe, 2022-08-12:
I have copied the source back into the main dir to get the project running.
 
[image]

Here I have tried three times to mill a small PCB. The ABL is not working...
Then I used bCNC with the Ant Farm G Code.  This time it works. ABL with bCNC. 
I have started the ABL process and have it activated.

Peter

TheAntTeam, 2022-08-19:
@dreschpe, it seems that there is an issue with the ABL on the bottom side that we are investigating. Are you trying to mill a bottom side?

dreschpe, 2022-08-19:
no ist was a single side front.

bvarner, 2022-08-22:
I had the same issues on the top side as well.

I started using bCNC and things work fine.

TheAntTeam, 2022-08-23:
Yes, it is possible. There is a bug that prevents ABL information from loading correctly. The bug can show up on both sides. We are trying to solve.

TheAntTeam, 2022-08-23:
Hi, can you try the fix? I have done some tests and It seems working

dreschpe, 2022-08-25:
I have pulled the actual code and it is working for a top layer.

Best Regards,
Peter