TheAntTeam/TheAntFarm #28: Feature request

dreschpe, 2022-08-12:
One problem is the setting of the milling depth. If you use different PCB materials you have to run tests.
Solution - you can adjust the milling depth in real time with two buttons + - 0.01 mm.
You start with a low milling depth. The bit is only scratching the copper. You carefully increase the milling depth and look at the result.
If the milled isolation is ok you press stop and restart the process. The repeat accuracy of the ant is good. The new traces correct the scratches lines from the beginning.

Peter

TheAntTeam, 2022-08-19:
Do you mean something like [Marlin Babystep](https://marlinfw.org/docs/gcode/M290.html)? (Implemented via software and not via firmware)

IckyDeh, 2022-08-23:
I support this request as this would be super helpful for solder mask removal with spring loaded bit.

TheAntTeam, 2022-08-23:
Yes, we also like this feature.  We just have to figure out how to insert it in a smart way.  Give us some time to think about it.