TheAntTeam/TheAntFarm #31: Install problem on Windows 10

tetref, 2022-10-11:
good day,
do you know which version of Python on WINDOWS works with the latest version of TheAntFarm?
Python 3.7 has a problem with the kiwisolver module, Python3.8 and anything higher has problems with pyside2.
I have Python 3.8 on Linux Mint and the program works fine...

TheAntTeam, 2022-10-20:
Hi,
thanks for your feedback.
I setup a virtual machine to re-try a clean install and I found the following:

1) you are right about python 3.7 and kiwisolver, in fact it's better not to use python 3.7 and I will modify the Readme to indicate to use python 3.8 and higher.
2) I tried a clean install with python 3.8.10 downloaded and installed from python.org, and I followed the Readme. The install succeeded and I was able to run the program.
3) I did not see problems with pyside2, but since my virtual machine does not support OpenGL, I have issue when using the application.

Could you give us a bit more details about your installation on windows with Python 3.8?
Which exact version did you use? Did you downloaded it from python site or from microsoft store? What is the commands you issued to do the installation? 
Also if you have a trace of the errors that would be really helpful. ;)

tetref, 2022-10-21:
Unfortunately, you don't have it described in the description that it doesn't work with the latest version of Python. So first I tried to install the program with the latest version of Python via the Microsoft Store.
I ended up installing Python 3.8.10 directly from the creator's site.
Here is my installation procedure on Windows 10 64bit:

cd C:\
git clone https://github.com/TheAntTeam/TheAntFarm.git 
cd TheAntFarm
python -V
_Python 3.8.10_
python -m venv .\env
.\env\Scripts\activate
python -m pip install --upgrade pip
pip -V
_pip 22.3 from c:\theantfarm\env\lib\site-packages\pip (python 3.8)_
pip install --no-cache-dir -r requirements.txt
pip list
_Package        Version
-------------- --------
cairocffi            0.9.0
cffi                    1.14.5
freetype-py        2.2.0
hsluv                  5.0.3
kiwisolver           1.3.2
numpy                1.22.3
opencv-python   4.6.0.66
pcb-tools            0.1.6
pip                      22.3
pyclipper            1.2.1
pycparser            2.20
PyOpenGL           3.1.5
PySide2               5.15.2.1
qimage2ndarray 1.8.3
scipy                    1.8.0
setuptools            56.0.0
Shapely                1.8.1
shiboken2            5.15.2.1
vispy                     0.9.4
wheel                    0.37.1_

cd .\src\TheAntFarm
python the_ant_farm.py

_Windows
Windows Env
WARNING: Unable to enumerate family ' "Organic Fridays Heart Alternates" '
WARNING: Unable to enumerate family ' "Organic Fridays Solid Alternates" '_

I finally managed to run the program but it is terribly slow, it only connects to some boards based on grbl or grblHAL but for example on a RaspberryPi Pico with grblHAL it connects but does not activate or respond to $$. But the same program on the same computer works perfectly fine in Linux.

TheAntTeam, 2022-10-26:
Ok, the installation issue is due almost certainly to the MS version of python that has issues with some packages installations.

I added a github action that builds The Ant Farm on every last version of the platform (windows included) with the python version from 3.8 to 3.10.

I close this issue about installation, for the other issue please open another one, and give us more details.

Thanks :)

closed 2022-10-26 by TheAntTeam